OpenNeptune3D/OpenNept4une #289: Weird additions to all entries in mainsail

Barkingmad15, 2025-05-02:
Hi all, completed the latest update as per instructions. I tend toprefer Mainsail over Fluidd but on Mainsail I now have weird additions to all the menus. Please see attached screenshot & also klippy.log (if that's helpful) These additions do not appear in Fluidd. The version of Open4neptune is: v0.1.5-556-ge844ceba & Klippper ver: v0.13.0-71-g3cf8899a

It's become unusable in Mainsail as I can't seem to adjust any temperatures etc. I have tried doing full system updates as well in SSH. I am loathe to try and roll back a version using Kiauh as I really really, don't want to have to take those darn screws all out the bottom again for a reflash, especially as I just got it settled in an enclosure!! Any advice would be gratefully recieived. Many thanks.

[klippy.log](https://github.com/user-attachments/files/20011826/klippy.log)
[image]

halfmanbear, 2025-05-05:
This looks unusual. You could try uninstall mainsail in kiauh, then reinstall mainsail in kiauh. Ensure you type 81 when asked which port it should run on.

Barkingmad15, 2025-05-06:
Thank-you for the suggestion. Have tried that & it just looks the same :(

halfmanbear, 2025-05-10:
It might be best raising the issue here https://github.com/mainsail-crew/mainsail/issues

Barkingmad15, 2025-05-12:
I was just heading over to follow your suggestion when a further software update showed as available. Thankfully, the update fixed it, but oddly enough it was an update to Klipper & not Mainsail. Thanks for the advice though.

closed 2025-05-12 by Barkingmad15