OpenNeptune3D/OpenNept4une #296: bord auswahl

Master0701, 2025-05-14:
Guten tag ist es beim einrichten auch möglich das bord v 2.0 auszuwelen wen es sich um ein neptun 4 pro handelt 
ich habe nemlich for mein n4pro von znp k1 v1.1 auf znp k1 v2.0 umzurüsten ????
wegen der treiber spanung frage ich denn im script steht was von 4 und 4pro mit treiber spanung ausweln und bord version 1.0 und 1.1 
da könte mann bestimt solte es nicht so sein das 2.0 bord in die aswahl mit aufnehmen ???

halfmanbear, 2025-05-14:
Hallo,
bitte laden Sie diese Datei (für das Plus/Max V2.0-Board) herunter und flashen Sie sie:

[Neptune4-Plus+Max-v0.1.5-ZNP-K1-V2.0.img.xz](https://github.com/OpenNeptune3D/OpenNept4une/releases/download/v0.1.5/Neptune4-Plus+Max-v0.1.5-ZNP-K1-V2.0.img.xz)

Während des Installationsvorgangs werden Sie nach Ihrem Maschinenmodell gefragt. Wählen Sie einfach „Neptune 4 Pro“ sowie den passenden Schrittmotorstrom aus.

Master0701, 2025-05-15:
ook denn geht das auch ok danke

closed 2025-05-16 by Master0701