OpenNeptune3D/OpenNept4une #302: Low on disk space after successful installation

tbnovak, 2025-05-22:
I had a successful conversion of my Neptune 4 Plus over to OpenNept4une. I'm getting it back up and running, and - in the Configuration section of Fluidd - I can see a yellow hand icon that tells me I'm low on disk space.

**What can I do after the installation to reclaim some space?**

I had a _mostly_ by-the-book install. I'll admit I deviated when I tried to install KlipperScreen with Kiauh. (It didn't work. My touchpad whined about an unsupported version. I resolved it by following the instructions for getting up-to-date with the touchpad version via SD Card. I'm digressing.) I also have a couple extra addons like OctoApp, OctoEverywhere, and Spoolman that are certainly taking up non-zero space. Crowsnest is gone because Kiauh guided me thru the mjpg-streamer. I don't know what else I can (safely) trim.

halfmanbear, 2025-05-22:
You can run this to clear some logs / cached files etc;

```sudo apt clean; sudo apt autoremove -y; sudo rm -rf /var/lib/apt/lists/*; rm -rf ~/.cache/pip; rm -f ~/printer_data/logs/*```

Previously uploaded/printed GCODE files taken up quite a bit of space over time. These can be deleted in fluidd or with;

```rm ~/printer_data/gcodes/*.gcode```

This command will show you the largest directories in your home folder although be careful when removing things, eg use kiauh to uninstall projects no longer required instead of removing dirs here.

```du -h ~ --max-depth=3 2>/dev/null | grep -Ev '(\.git|/lib/|/include)' | sort -hr | head -n 30```

This command will find the largest files taking up disk space throughout the whole system;

```sudo find / -xdev -type f -size +10M ! -path "/proc/*" ! -path "/sys/*" ! -path "/dev/*" ! -path "/boot/*" ! -path "/run/*" ! -path "/var/log/*" -exec du -h {} + 2>/dev/null | sort -rh | head -n 50```

plygrnd, 2025-05-25:
This should be a cron, imo:

```bash
#!/bin/bash

# run as root

# clear apt artifacts
apt clean
apt autoremove -y
rm -rf /var/lib/apt/lists/* 

# clear pip cache
rm -rf ~/.cache/pip; 

# purge old log files
rm -f ~/printer_data/logs/*
```

tbnovak, 2025-05-25:
I ran the cleanup commands and rebooted. The hand is gone. Thank you so much.

Output from the script to find big files (after the cleanup was done):
```
153M    /home/mks/klipper/.git/objects/pack/pack-2ffe73dc74560dfaf9713a853ec34cff9865c5e7.pack
107M    /usr/lib/aarch64-linux-gnu/libLLVM-15.so.1
75M     /usr/lib/firmware/qcom/sc8180x/SAMSUNG/W767/qcmpss8180_XEF.mbn
71M     /home/mks/klipper/.git/objects/pack/pack-15a14cd39467158c0e3bf51ee2d7ca6356e73067.pack
40M     /home/mks/printer_data/gcodes/ConeFidget_0.1mm_PLA_Elegoo Neptune 4 Plus_11h46m.gcode
39M     /var/cache/apt/srcpkgcache.bin
39M     /var/cache/apt/pkgcache.bin
34M     /usr/lib/aarch64-linux-gnu/openblas-pthread/libopenblasp-r0.3.21.a
30M     /usr/lib/aarch64-linux-gnu/libicudata.so.72.1
30M     /usr/lib/aarch64-linux-gnu/libicudata.a
29M     /usr/lib/gcc/aarch64-linux-gnu/12/cc1plus
27M     /usr/lib/gcc/arm-none-eabi/12.2.1/cc1plus
27M     /usr/lib/gcc/aarch64-linux-gnu/12/cc1
26M     /usr/lib/gcc/aarch64-linux-gnu/12/lto1
26M     /usr/bin/aarch64-linux-gnu-lto-dump-12
26M     /home/mks/Spoolman/.venv/bin/ruff
25M     /usr/lib/gcc/arm-none-eabi/12.2.1/cc1
25M     /home/mks/klippy-env/lib/python3.11/site-packages/numpy.libs/libopenblas64_p-r0-17488984.3.23.dev.so
24M     /usr/bin/arm-none-eabi-lto-dump
24M     /home/mks/display_connector/venv/lib/python3.11/site-packages/numpy.libs/libscipy_openblas64_-128b20d9.so
23M     /usr/lib/gcc/arm-none-eabi/12.2.1/lto1
23M     /usr/lib/aarch64-linux-gnu/dri/armada-drm_dri.so
22M     /usr/lib/aarch64-linux-gnu/libz3.so.4
21M     /home/mks/printer_data/gcodes/Octopus_spiral_v6_0.1mm_PLA_Elegoo Neptune 4 Plus_3h30m.gcode
21M     /home/mks/octoapp/.git/objects/pack/pack-57e45cc47b731ed36b58057e5ae330b6a8b35ffc.pack
20M     /usr/lib/aarch64-linux-gnu/openblas-pthread/libopenblasp-r0.3.21.so
18M     /usr/lib/firmware/qcom/sm8250/xiaomi/elish/adsp.mbn
18M     /usr/lib/aarch64-linux-gnu/libavcodec.a
17M     /usr/lib/aarch64-linux-gnu/libcodec2.so.1.0
15M     /home/mks/Klipper-Adaptive-Meshing-Purging/.git/objects/pack/pack-0c6137a51aa34bd7ce0794550b395d78a7882ac8.pack
14M     /usr/lib/firmware/qcom/sc8280xp/MICROSOFT/DEVKIT23/qcadsp8280.mbn
14M     /home/mks/Spoolman/.venv/lib/python3.11/site-packages/uvloop/loop.cpython-311-aarch64-linux-gnu.so
13M     /usr/lib/udev/hwdb.bin
13M     /usr/lib/gcc/avr/5.4.0/cc1plus
12M     /usr/lib/python3/dist-packages/sympy/integrals/rubi/rubi_tests/tests/__pycache__/test_trinomials.cpython-311.pyc
12M     /usr/lib/python3.11/config-3.11-aarch64-linux-gnu/libpython3.11-pic.a
12M     /usr/lib/python3.11/config-3.11-aarch64-linux-gnu/libpython3.11.a
12M     /usr/lib/gcc/avr/5.4.0/cc1
12M     /usr/lib/aarch64-linux-gnu/libavcodec.so.59.37.100
12M     /usr/lib/aarch64-linux-gnu/atlas/liblapack.a
12M     /home/mks/Klipper-Adaptive-Meshing-Purging/Photos/Purging-Assets/voron-purge-example.png
12M     /home/mks/display_connector/dev-resources/UI_1.2.14_240705.tft
12M     /home/mks/display_connector/dev-resources/ui_1.2.11_231012.tft
11M     /usr/lib/gcc/avr/5.4.0/lto1
11M     /usr/lib/firmware/qcom/sc8180x/SAMSUNG/W767/qcadsp8180.mbn
11M     /usr/lib/aarch64-linux-gnu/libavfilter.so.8.44.100
11M     /home/mks/octoeverywhere-env/lib/python3.11/site-packages/zstandard/_cffi.cpython-311-aarch64-linux-gnu.so
11M     /etc/oh-my-zsh/.git/objects/pack/pack-4e9fc0355eb464753e1b093aca22e272088a427d.pack
```
So, Klipper, Arch Linux lib, device firmware, Klipper again, then one of my gcodes are top 5. I'm not seeing anything else to clean up, but if something catches your eye, I'd love to get even leaner. It'd be nice to see these commands in the installation wiki, but my immediate issues have been resolved.

closed 2025-05-25 by tbnovak