OpenNeptune3D/OpenNept4une #306: Default retraction=0 (N4P)

tbnovak, 2025-05-27:
The Neptune 4 Plus config doesn't have a setting for [firmware_retraction], best I can tell. The section exists, but there's no data. "Okay, fine, the slicer can take care of it."
Except the Orca profile for the Neptune 4 Plus says to use firmware retraction.

I just spent a butt-load of time trying to figure out why my prints got so hairy after the switch to Open, which coincided with the switch to Orca. I'd heard that the profiles were great, but this was a big let down. Recommend fixing one way or the other; I really don't know what the "standard" is. Change the Orca profile to not use firmware retraction. Or change the config to include [firmware_retraction] values.

halfmanbear, 2025-05-27:
We have firmware retraction configured in the filament profiles as per the documentation. We did this because each filament type typically requires it's own retraction tuning. 

https://github.com/OpenNeptune3D/OpenNept4une/wiki/Printer-Calibration-%E2%80%90-Klipper-&-OrcaSlicer#firmware-retraction-values

tbnovak, 2025-05-27:
Okay, that makes sense why it's missing then. I can't find the filament profiles. I found a machine profile, a print profile, and that's it. Nothing for filament.

Can you point me to that?

tbnovak, 2025-05-29:
No, I'm mistaken. I've mixed Cura and Orca behavior. Cura has retraction set in the "Process" (or its equivalent). Un-checking the "Use Firmware Retraction" box goes to the Printer > Extruder setting. Which is definitely not per-filament. I stand very corrected. It sounds like the missing filament profiles are the intended solution.

halfmanbear, 2025-05-29:
Can you rename the downloaded orca profiles file extension to a .zip extension and extract it.

It should contain everything there, I'm unable to check right now.

tbnovak, 2025-05-30:
Changed to .zip and I can see that there's a Filament folder, but Orca 2.3.0 isn't recognizing them. I declined to override the 3 configs I already imported and tweaked so it concluded with "There are 0 configs imported. (Only non-system and compatible configs). Hint: Make sure you have added the corresponding printer before importing the configs."

halfmanbear, 2025-05-31:
Thankyou for confirming. I'll investigate what's wrong with those profiles.

halfmanbear, 2025-06-02:
I've found the issue, for some reason OrcaSlicer has renamed the original Elegoo profiles. 

These two should work for now.

[Elegoo Generic PETG-OpenNept4une.json](https://github.com/user-attachments/files/20551924/Elegoo.Generic.PETG-OpenNept4une.json)
[Elegoo Generic PLA-OpenNept4une.json](https://github.com/user-attachments/files/20551923/Elegoo.Generic.PLA-OpenNept4une.json)
 

[image]

halfmanbear, 2025-06-02:
I have updated all OrcaSlicer profiles in the dev branch, below is the full N4Plus Orca profile bundle

https://github.com/OpenNeptune3D/OpenNept4une/raw/refs/heads/dev/orca-profiles/n4plus/Elegoo%20Neptune%204%20Plus%20(0.4%20nozzle)-OpenNept4une.orca_printer